Flipper-XFW/Xtreme-Firmware #508: NFC maker App

LePtitPhiloup, 2024-01-03:
### Describe the bug.

Hi 

When i'm creating a nfc tag for Wifi login and try to emulate it directly from Flipper zero, phone don't connect to the wifi but open a screen "new tag added" and it's wrote "application/vnd.wfa.wsc and don't connect phones to my wifi 

hope you know why. maybe i do something wrong, If so please let me know

thanks for all your work
happy new year !
Phil

### Reproduction

1 choose a wifi access point and find SSID, password, Authentification type and encryption type (in my case I took my android shared access point)
2 power up your flipper zero and look for NFC maker app
3 go down and start makin a "Wifi login" tag
4 fill the form with data
5 save the tag
6 go to NFC => saved => NewTagName => Emulate
7 put the back of flipper zero to the back of your phone

LePtitPhiloup, 2024-01-03:
Hi thank you for answering me
I'm just a bit disapointed you don't even try to understand because I made many tests without multiple wifi and phones
Is a Pixel 7 pro not good enought ?

Willy-JL, 2024-01-13:
send the file you made

Willy-JL, 2024-02-07:
i was working on other stuff and happened to rework the wifi generator code, noticed an error in it. i hope this nuisance is what was causing some devices to not recognize it correctly.

@LePtitPhiloup try it on the latest dev build and see if it works now. if it does not, then it truly is just a problem of your device's implementation, since the nfc maker now generates dumps that are 100% identical to the ones made by all "nfc tools" and similar apps on phone.

anickname028, 2024-02-11:
You could also try making a NTAG-216 from the NFC (not NFC Maker) app and writing the WiFi record using a app like NFC Tools.

LePtitPhiloup, 2024-02-13:
Hello
Thanks @Willy-JL  for noticing me about your work and thanks for your work. I'll give it a try asap !
Thanks also @anickname028 for your advice !
Have a good day !