Flipper-XFW/Xtreme-Firmware #514: Sub-GHz emulation: Quickly tapping Send doesn't reliably send data packet

walbergs, 2024-01-09:
### Describe the bug.

During Sub-GHz emulation, quickly tapping Send doesn't reliably send the data packet; the Send button needs to be held down. I've tested this with a garage door opener and a motion detector chime.

When using rolling codes, this results in the key advancing on the Flipper without the receiver (i.e. Garage door opener) receiving it.

### Reproduction

1. Emulate a Sub-GHz signal.
2. Tap Send button (quickly).
3. Nothing happens on receiver.
4. Hold Send button.
5. Receiver receives signal.

ALEEF02, 2024-01-19:
+1, I've personally dealt with this and would like to see a fix implemented.

MatthewKuKanich, 2024-02-22:
To add, this seems to be an issue with sending saved protocols. I haven't experienced this when emulating raw signals, but it will behave as described when sending Security+ 1.0 & 2.0 signals. The flipper considers the unsent code as "used" so the values increment despite no signal actually being sent. You can even rapidly press the send button to advance the values several times without sending any signal. The seed updating logic seems to be performed prior to sending the signal.