Flipper-XFW/Xtreme-Firmware #517: Timestamp on SubGhz files conflicting

geckom, 2024-01-12:
### Description of the feature you're suggesting.

Currently when you save a SubGhz file with a timestamp filename you get something like this "SubGHz_2024-01-12_09,43.sub"

Since the resolution is only down to the minute, if you try to save 2 captured within the same minute you will need to rename the second file to something different.

Proposed solution - add a seconds number to the file name ie. SubGHz_2024-01-12_09,43,42.sub

smart734, 2024-01-23:
Hi, I also encountered this problem when I had the "Bin RAW" function of the Sub-GHZ turned on. I'm not using this at the moment, but switching it in "Settings - System - File Naming - Random" might help. Try it

Willy-JL, 2024-01-24:
addressed this in 461026ee27078d92e6d3c1a789a37f006db4a15b, it now includes seconds too.

also, i noticed that subghz used the time of you pressing "save" for the timestamp. added a new api in 7e7509d48177b2593152d12d096206ae12a63ce6 to generate filenames with custom timestamp, and fixed subghz to use it in a1c7dc5eaa13905fa3282f7338408756b7af8cb0. now subghz uses timestamp of signal receival.

also, with 411a65a42e3dde13dd037c1a6d00130892246f60 you can choose to have timestamps before the protocol name, which can be useful for sorting.

closed 2024-02-02 by Willy-JL