Flipper-XFW/Xtreme-Firmware #522: ble_spam app fail XFW-0052_09122023

AIArt20, 2024-01-15:
### Describe the bug.

The flipper ble_spam app fails in XFW-0052_09122023 or Apple/Android devices have been updated.

### Reproduction

Using the ble_spam app XFW-0052_09122023.

Willy-JL, 2024-01-15:
did you expect it to work forever and apple/google/samsung/microsoft to never fix it? lol

will still work for some popups, but less most likely

closed 2024-01-15 by Willy-JL

AIArt20, 2024-01-16:
this app than can be removed for no future use? lol

Willy-JL, 2024-01-16:
I literally told you that it still works. Just less effective.

Willy-JL, 2024-01-16:
No wonder such stupid logic comes from "AiArt20"

AIArt20, 2024-01-16:
Less effective... lol, Id rather call it obsolete. But ok, ok, let's keep these nonsense.

Willy-JL, 2024-01-16:
If you can't get it to work, you are the one less effective at using this "nonsense". Learn to read basic instructions, and maybe activate the 2 last remaining neurons in your brain