Flipper-XFW/Xtreme-Firmware #525: But on mifaremagic - not present on official firmware

MarianoFacundoArch, 2024-01-21:
### Describe the bug.

the app mifare magic on xtreme firmware is bugged. It cannot write a gen4 mifare classic 4k. It writes as a mifare mini and fails. 
I even noticed several crashes and misbehavior. 
I installed the latest dev official firmware and removed xtreme and it works perfectly. 

### Reproduction

Clone a mifare classic 4k
Write it to a 4 gen card

Willy-JL, 2024-01-21:
How about you install the latest xtreme dev and see if it's fixed?

Willy-JL, 2024-02-02:
Stale, and I assume this is fixed. Feel free to post an update if you still have issues

closed 2024-02-02 by Willy-JL