Flipper-XFW/Xtreme-Firmware #526: NFC writing through NFC Magic is entirely broken in current release

koolfy, 2024-01-21:
### Describe the bug.

During some tests with a Ultimate Magic card (gen4), I noticed any attempt at writing data to the card through the NFC Magic app failed.
The "Wipe Card" failed as well, and had an apparently random chance of soft-bricking the card.
At that point, the card is completely useless unless you get a proxmark3 to issue a proper reset command.

I wanted to dig further, but simply by switching over to the latest upstream release of 0.97.1 fixed all these issues.

There were a bunch of refactoring and tweaks to the NFC stack upstream, so my theory is the current Xtreme build got caught in the middle of some important ones, and they have since been fixed.

This issue is simply a warning that the current latest build has some broken functionalities, that will simply go away with the next Xtreme release when you ship the upstream fixes.

I don't know if this is worth updating the "known issues" section.

Basically, don't spent too much time investigating this, it will sort itself out :)

### Reproduction

1. Install NFC Magic (latest) on the flipper with Xtreme firmware
2. Check tag of a Ultimate Magic card (gen4)
3. Attempt any write operation (write a saved NFC dump or try wiping the card
4. Notice that writes fail and wiping the card a few times results in a soft-brick (currently unrecoverable with the current NFC Magic and thus requiring dedicated hardware)

### Logs

```Text
N/A this problem is not worth an in-depth investigation, it is already fixed upstream
```

### Anything else?

This is a placeholder issue, the problem is already fixed upstream and very likely will disappear with the next official release (this might be a good reason to push one out the door soon?)

koolfy, 2024-01-21:
In case people are curious, NFC Magic cannot currently reset a soft-bricked ultimate magic card (gen4), because such a card fails the initial "check tag" operation, which gates the wipe operation.

I'm thinking of doing a contribution to improve this in the app, but in the meantime, **don't attempt repeated "wipes" through NFC magic on the current XFW-0052_09122023  build if you don't have access to a Proxmark3 device**

FYI if you do, this will reset the card back to a usable state:
`hf 14a raw -s -c -t 1000 CF00000000F000000000000002000978009102DABC19101011121314151604000800`

Willy-JL, 2024-01-21:
Install latest xtreme dev and see if it's fixed

koolfy, 2024-01-21:
I wanted to try this, but couldn't find instructions on how to build/flash the latest head of dev for Xtreme.
Maybe it's obvious, sorry, this ecosystem is still new to me :)

koolfy, 2024-01-21:
Seems fixed when building nad installing latest DEV (b833ebab1a3cea883aa53c2afa6f08befb367f3c)
There is a problem with the "wiping" operation but other write operations seem to work fine so I suspect something changed with NFC Magic and will investigate on the app side :)

Doesn't look like a firmware-wide issue anymore!

Willy-JL, 2024-01-21:
As you said this is most likely an issue on app side. All depends how you install the app and what version. With xtreme you get the app version from out apps repo, it's updated every few days, but there might be new changes upstream in official nfc magic repo. As for official firmware, I'm guessing you are installing the app from app hub, which is stable releases rather than development versions. We always pull the latest from upstream apps

Willy-JL, 2024-01-21:
That's to say maybe it's a bug that is being actively developed and fixed, and isn't quite there yet, not sure. But if you install the same app on xtreme and ofw and get different results, then that is an issue that should be brought up to us. Or if the issue is fixed upstream and we are yet to include the fix, then that too.

koolfy, 2024-01-21:
I'll try to determine what introduced/fixed the issue currently affecting the latest release of Xtreme, before we close this issue if possible!

closed 2024-01-22 by koolfy

Willy-JL, 2024-01-22:
Thanks for taking the time and experimenting with it, most wouldn't and would just demand a fix with no further explanation or research. Props! And yeah release will be not too far from now I hope