Flipper-XFW/Xtreme-Firmware #531: Mouse jiggler not working Win 11

Mfd81, 2024-01-24:
### Describe the bug.

Mouse Jiggler not working on windows 11 over usb.

### Reproduction

1. use the mouse jiggler on windows 11

### Target

windows 11 pc

Willy-JL, 2024-01-24:
Are you using the usb remote app? If so, do the other options work in it?

Mfd81, 2024-01-24:
yes, and yes the mouse clicker and everything else works.

Mfd81, 2024-01-24:
Thank You for the quick reply, Love the firmware too!

Willy-JL, 2024-01-24:
It did change yes. I'll change it back after hearing this tho

spralexx, 2024-01-25:
Can confirm. Getting logged of in Citrix. Has been working before.

Willy-JL, 2024-01-28:
actually, nevermind. in the past it was 5 pixels jiggle. then with app migration and updates from upstream it became 1 pixel in the new apps repo, this is how it is in v52. but it has already been changed back to 5 pixels in dev branch. use the latest devbuild and let me know if it works, but it should.

closed 2024-02-02 by Willy-JL