Flipper-XFW/Xtreme-Firmware #533: azerty keyboard

KaliStudio, 2024-01-26:
### Description of the feature you're suggesting.

The possibility of putting the keyboard in azerty because for example the bluetooth keyboard gives bad results on windows, if I press Q on the flipper zero, it displays A on my windows.
Sorry, French keyboard

ClaraCrazy, 2024-01-26:
you literally can in the settings...

KaliStudio, 2024-01-26:
Can you explain to me how as I can't find the settings for app\bluetooth\bluetooth Remote\Keyboard

KaliStudio, 2024-01-26:
I tried to modify the file C:\Xtreme-Firmware\applications\system\hid_app\views\hid_keyboard.c but the characters éèçà¨£§ùµ° are not displayed 

```
#define MARGIN_TOP 0
#define MARGIN_LEFT 4
#define KEY_WIDTH 9
#define KEY_HEIGHT 12
#define KEY_PADDING 1
#define ROW_COUNT 7
#define COLUMN_COUNT 14

// 0 width items are not drawn, but there value is used
const HidKeyboardKey hid_keyboard_keyset[ROW_COUNT][COLUMN_COUNT] = {
    {
        {.width = 1, .icon = &I_ButtonF1_5x8, .value = HID_KEYBOARD_F1},
        {.width = 1, .icon = &I_ButtonF2_5x8, .value = HID_KEYBOARD_F2},
        {.width = 1, .icon = &I_ButtonF3_5x8, .value = HID_KEYBOARD_F3},
        {.width = 1, .icon = &I_ButtonF4_5x8, .value = HID_KEYBOARD_F4},
        {.width = 1, .icon = &I_ButtonF5_5x8, .value = HID_KEYBOARD_F5},
        {.width = 1, .icon = &I_ButtonF6_5x8, .value = HID_KEYBOARD_F6},
        {.width = 1, .icon = &I_ButtonF7_5x8, .value = HID_KEYBOARD_F7},
        {.width = 1, .icon = &I_ButtonF8_5x8, .value = HID_KEYBOARD_F8},
        {.width = 1, .icon = &I_ButtonF9_5x8, .value = HID_KEYBOARD_F9},
        {.width = 1, .icon = &I_ButtonF10_5x8, .value = HID_KEYBOARD_F10},
        {.width = 1, .icon = &I_ButtonF11_5x8, .value = HID_KEYBOARD_F11},
        {.width = 1, .icon = &I_ButtonF12_5x8, .value = HID_KEYBOARD_F12}, //12
    },
    {
        {.width = 1, .icon = NULL, .key = "&", .shift_key = "1", .value = HID_KEYBOARD_1},
        {.width = 1, .icon = NULL, .key = "é", .shift_key = "2", .value = HID_KEYBOARD_2}, //é ne s'affiche pas fonctionne pas : "\u00E9" &eacute; "\xE9" "Alt+130"
        {.width = 1, .icon = NULL, .key = "\"", .shift_key = "3", .value = HID_KEYBOARD_3},
        {.width = 1, .icon = NULL, .key = "'", .shift_key = "4", .value = HID_KEYBOARD_4},
        {.width = 1, .icon = NULL, .key = "(", .shift_key = "5", .value = HID_KEYBOARD_5},
        {.width = 1, .icon = NULL, .key = "-", .shift_key = "6", .value = HID_KEYBOARD_6},
        {.width = 1, .icon = NULL, .key = "è", .shift_key = "7", .value = HID_KEYBOARD_7}, //è ne s'affiche pas 
        {.width = 1, .icon = NULL, .key = "_", .shift_key = "8", .value = HID_KEYBOARD_8},
        {.width = 1, .icon = NULL, .key = "ç", .shift_key = "9", .value = HID_KEYBOARD_9}, //ç ne s'affiche pas 
        {.width = 1, .icon = NULL, .key = "à", .shift_key = "0", .value = HID_KEYBOARD_0}, //à ne s'affiche pas 	
        {.width = 2, .icon = &I_Pin_arrow_left_9x7, .value = HID_KEYBOARD_DELETE},
        {.width = 0, .value = HID_KEYBOARD_DELETE}, //12
    },
    {
	    {.width = 1, .icon = NULL, .key = "a", .shift_key = "A", .value = HID_KEYBOARD_Q},
        {.width = 1, .icon = NULL, .key = "z", .shift_key = "Z", .value = HID_KEYBOARD_W},
        {.width = 1, .icon = NULL, .key = "e", .shift_key = "E", .value = HID_KEYBOARD_E},
        {.width = 1, .icon = NULL, .key = "r", .shift_key = "R", .value = HID_KEYBOARD_R},
        {.width = 1, .icon = NULL, .key = "t", .shift_key = "T", .value = HID_KEYBOARD_T},
        {.width = 1, .icon = NULL, .key = "y", .shift_key = "Y", .value = HID_KEYBOARD_Y},	
        {.width = 1, .icon = NULL, .key = "u", .shift_key = "U", .value = HID_KEYBOARD_U},
        {.width = 1, .icon = NULL, .key = "i", .shift_key = "I", .value = HID_KEYBOARD_I},
        {.width = 1, .icon = NULL, .key = "o", .shift_key = "O", .value = HID_KEYBOARD_O},
        {.width = 1, .icon = NULL, .key = "p", .shift_key = "P", .value = HID_KEYBOARD_P},	
        {.width = 1, .icon = NULL, .key = "^", .shift_key = "¨", .value = HID_KEYBOARD_OPEN_BRACKET},	 //¨ ne s'affiche pas 
        {.width = 1, .icon = NULL, .key = "$", .shift_key = "£", .value = HID_KEYBOARD_CLOSE_BRACKET},	 //£ ne s'affiche pas 	//12
    },	
    {
        {.width = 1, .icon = NULL, .key = "q", .shift_key = "Q", .value = HID_KEYBOARD_A},
        {.width = 1, .icon = NULL, .key = "s", .shift_key = "S", .value = HID_KEYBOARD_S},
        {.width = 1, .icon = NULL, .key = "d", .shift_key = "D", .value = HID_KEYBOARD_D},
        {.width = 1, .icon = NULL, .key = "f", .shift_key = "F", .value = HID_KEYBOARD_F},
        {.width = 1, .icon = NULL, .key = "g", .shift_key = "G", .value = HID_KEYBOARD_G},
        {.width = 1, .icon = NULL, .key = "h", .shift_key = "H", .value = HID_KEYBOARD_H},
        {.width = 1, .icon = NULL, .key = "j", .shift_key = "J", .value = HID_KEYBOARD_J},
        {.width = 1, .icon = NULL, .key = "k", .shift_key = "K", .value = HID_KEYBOARD_K},
        {.width = 1, .icon = NULL, .key = "l", .shift_key = "L", .value = HID_KEYBOARD_L},
        {.width = 1, .icon = NULL, .key = "m", .shift_key = "M", .value = HID_KEYBOARD_SEMICOLON}, 
		{.width = 2, .icon = &I_Pin_arrow_right_9x7, .value = HID_KEYBOARD_RETURN}, 
        {.width = 0, .value = HID_KEYBOARD_RETURN}, //12
    },
    {
[... 17 lines cut ...]
        {.width = 0, .value = HID_KEYBOARD_SPACEBAR},
        {.width = 0, .value = HID_KEYBOARD_SPACEBAR},
        {.width = 0, .value = HID_KEYBOARD_SPACEBAR},
		{.width = 1, .icon = NULL, .key = ")", .shift_key = "°", .value = HID_KEYBOARD_MINUS}, //° ne s'affiche pas 
		{.width = 1, .icon = NULL, .key = "=", .shift_key = "+", .value = HID_KEYBOARD_EQUAL_SIGN},	
        {.width = 1, .icon = &I_ButtonLeft_4x7, .value = HID_KEYBOARD_LEFT_ARROW},
        {.width = 1, .icon = &I_ButtonDown_7x4, .value = HID_KEYBOARD_DOWN_ARROW},
        {.width = 1, .icon = &I_ButtonRight_4x7, .value = HID_KEYBOARD_RIGHT_ARROW}, //11
    },
    {
[... cut ...]
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_L_GUI},
        {.width = 2, .icon = &I_KB_key_Tab_17x10, .value = HID_KEYBOARD_TAB},
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_TAB},
        {.width = 2, .icon = &I_KB_key_Esc_17x10, .value = HID_KEYBOARD_ESCAPE},
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_ESCAPE},
        {.width = 2, .icon = &I_KB_key_Del_17x10, .value = HID_KEYBOARD_DELETE_FORWARD},
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_DELETE_FORWARD},//12
    },
};
```

Willy-JL, 2024-01-26:
There is no setting for it in bt and usb remote, because those apps have a small usefulness enough that it does not make sense to implement this feature. Also, utf8 is not supported and will never be on flipper, so no point.

closed 2024-01-26 by Willy-JL

KaliStudio, 2024-01-26:
> Il n'y a aucun paramètre pour cela dans BT et USB Remote, car ces applications ont une utilité suffisamment petite pour qu'il n'ait pas de sens d'implémenter cette fonctionnalité. De plus, utf8 n'est pas pris en charge et ne sera jamais sur flipper, donc cela ne sert à rien.

Okay, thank you for your answer, can you explain to me if possible how I can use images instead of special characters like for arrows for example please? 

```
        {.width = 1, .icon = &I_ButtonLeft_4x7, .value = HID_KEYBOARD_LEFT_ARROW},
        {.width = 1, .icon = &I_ButtonDown_7x4, .value = HID_KEYBOARD_DOWN_ARROW},
        {.width = 1, .icon = &I_ButtonRight_4x7, .value = HID_KEYBOARD_RIGHT_ARROW}, 
```

KaliStudio, 2024-01-26:
[image]
[image]

modify the file Xtreme-Firmware\applications\system\hid_app\views\hid_keyboard.c
```
#define MARGIN_TOP 0
#define MARGIN_LEFT 4
#define KEY_WIDTH 9
#define KEY_HEIGHT 12
#define KEY_PADDING 1
#define ROW_COUNT 7
#define COLUMN_COUNT 14

// 0 width items are not drawn, but there value is used
const HidKeyboardKey hid_keyboard_keyset[ROW_COUNT][COLUMN_COUNT] = {
    {
        {.width = 1, .icon = &I_ButtonF1_5x8, .value = HID_KEYBOARD_F1},
        {.width = 1, .icon = &I_ButtonF2_5x8, .value = HID_KEYBOARD_F2},
        {.width = 1, .icon = &I_ButtonF3_5x8, .value = HID_KEYBOARD_F3},
        {.width = 1, .icon = &I_ButtonF4_5x8, .value = HID_KEYBOARD_F4},
        {.width = 1, .icon = &I_ButtonF5_5x8, .value = HID_KEYBOARD_F5},
        {.width = 1, .icon = &I_ButtonF6_5x8, .value = HID_KEYBOARD_F6},
        {.width = 1, .icon = &I_ButtonF7_5x8, .value = HID_KEYBOARD_F7},
        {.width = 1, .icon = &I_ButtonF8_5x8, .value = HID_KEYBOARD_F8},
        {.width = 1, .icon = &I_ButtonF9_5x8, .value = HID_KEYBOARD_F9},
        {.width = 1, .icon = &I_ButtonF10_5x8, .value = HID_KEYBOARD_F10},
        {.width = 1, .icon = &I_ButtonF11_5x8, .value = HID_KEYBOARD_F11},
        {.width = 1, .icon = &I_ButtonF12_5x8, .value = HID_KEYBOARD_F12},
    },
    {
        {.width = 1, .icon = NULL, .key = "&", .shift_key = "1", .value = HID_KEYBOARD_1},
		{.width = 1, .icon = &I_Buttoneaigu_7x10, .key = "é", .shift_key = "2", .value = HID_KEYBOARD_2},
        {.width = 1, .icon = NULL, .key = "\"", .shift_key = "3", .value = HID_KEYBOARD_3},
        {.width = 1, .icon = NULL, .key = "'", .shift_key = "4", .value = HID_KEYBOARD_4},
        {.width = 1, .icon = NULL, .key = "(", .shift_key = "5", .value = HID_KEYBOARD_5},
        {.width = 1, .icon = NULL, .key = "-", .shift_key = "6", .value = HID_KEYBOARD_6},
        {.width = 1, .icon = &I_Buttonegrave_7x10, .key = "è", .shift_key = "7", .value = HID_KEYBOARD_7}, 
        {.width = 1, .icon = NULL, .key = "_", .shift_key = "8", .value = HID_KEYBOARD_8},
        {.width = 1, .icon = &I_Buttoncced_7x10, .key = "ç", .shift_key = "9", .value = HID_KEYBOARD_9}, 
        {.width = 1, .icon = &I_Buttonagrave_7x10, .key = "à", .shift_key = "0", .value = HID_KEYBOARD_0}, 	
        {.width = 2, .icon = &I_Pin_arrow_left_9x7, .value = HID_KEYBOARD_DELETE},
        {.width = 0, .value = HID_KEYBOARD_DELETE}, 
    },
    {
	    {.width = 1, .icon = NULL, .key = "a", .shift_key = "A", .value = HID_KEYBOARD_Q},
        {.width = 1, .icon = NULL, .key = "z", .shift_key = "Z", .value = HID_KEYBOARD_W},
        {.width = 1, .icon = NULL, .key = "e", .shift_key = "E", .value = HID_KEYBOARD_E},
        {.width = 1, .icon = NULL, .key = "r", .shift_key = "R", .value = HID_KEYBOARD_R},
        {.width = 1, .icon = NULL, .key = "t", .shift_key = "T", .value = HID_KEYBOARD_T},
        {.width = 1, .icon = NULL, .key = "y", .shift_key = "Y", .value = HID_KEYBOARD_Y},	
        {.width = 1, .icon = NULL, .key = "u", .shift_key = "U", .value = HID_KEYBOARD_U},
        {.width = 1, .icon = NULL, .key = "i", .shift_key = "I", .value = HID_KEYBOARD_I},
        {.width = 1, .icon = NULL, .key = "o", .shift_key = "O", .value = HID_KEYBOARD_O},
        {.width = 1, .icon = NULL, .key = "p", .shift_key = "P", .value = HID_KEYBOARD_P},	
		{.width = 1, .icon = &I_Buttontrema_7x10, .key = "^", .shift_key = "¨", .value = HID_KEYBOARD_OPEN_BRACKET},	 //¨ ne s'affiche pas 
        {.width = 1, .icon = &I_ButtonLivreSterling_7x10, .key = "$", .shift_key = "£", .value = HID_KEYBOARD_CLOSE_BRACKET},	 //£ ne s'affiche pas 	//12
    },	
    {
        {.width = 1, .icon = NULL, .key = "q", .shift_key = "Q", .value = HID_KEYBOARD_A},
        {.width = 1, .icon = NULL, .key = "s", .shift_key = "S", .value = HID_KEYBOARD_S},
        {.width = 1, .icon = NULL, .key = "d", .shift_key = "D", .value = HID_KEYBOARD_D},
        {.width = 1, .icon = NULL, .key = "f", .shift_key = "F", .value = HID_KEYBOARD_F},
        {.width = 1, .icon = NULL, .key = "g", .shift_key = "G", .value = HID_KEYBOARD_G},
        {.width = 1, .icon = NULL, .key = "h", .shift_key = "H", .value = HID_KEYBOARD_H},
        {.width = 1, .icon = NULL, .key = "j", .shift_key = "J", .value = HID_KEYBOARD_J},
        {.width = 1, .icon = NULL, .key = "k", .shift_key = "K", .value = HID_KEYBOARD_K},
        {.width = 1, .icon = NULL, .key = "l", .shift_key = "L", .value = HID_KEYBOARD_L},
        {.width = 1, .icon = NULL, .key = "m", .shift_key = "M", .value = HID_KEYBOARD_SEMICOLON}, 
		{.width = 2, .icon = &I_Pin_arrow_right_9x7, .value = HID_KEYBOARD_RETURN}, 
        {.width = 0, .value = HID_KEYBOARD_RETURN}, 
[... 126 lines cut ...]
        }
        canvas_draw_str_aligned(
            canvas,
            MARGIN_LEFT + x * (KEY_WIDTH + KEY_PADDING) + keyWidth / 2 + 1,
            MARGIN_TOP + y * (KEY_HEIGHT + KEY_PADDING) + KEY_HEIGHT / 2,
            AlignCenter,
            AlignCenter,
            model->key_string);
    }
}
```

and added images in Xtreme-Firmware\applications\system\hid_app\assets : 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

then compile : ./fbt COMPACT=1 DEBUG=0 updater_package