Flipper-XFW/Xtreme-Firmware #534: Add UTF-8 support

KaliStudio, 2024-01-26:
### Description of the feature you're suggesting.

I haven't familiarized myself enough with pinball yet to do it myself but UTF-8 support would be great

Willy-JL, 2024-01-26:
You will not do it yourself, period. Supporting utf8 everywhere means fundamentally rewriting the entire firmware because it all excpects 1 byte = 1 character. Change to Unicode, now each character is sometimes 1 byte sometimes 2 bytes. It all falls apart. And there is no real usefulness to doing this anyway. And the fonts would need to include all the characters too, which go from 100 characters with ascii to 9999 characters with Unicode. There is already not enough space on flipper flash, this would not work. Flipper has 1mb flash that is already full, and 256kb ram that is already a problem. This will never happen. Not on flipper zero.

closed 2024-01-26 by Willy-JL

KaliStudio, 2024-01-28:
by modifying this line: case FontKeyboard:
       //u8g2_SetFont(&canvas->fb, u8g2_font_profont11_mr);
         u8g2_SetFont(&canvas->fb, u8g2_font_chikita_tf); //utf8
         break;

my characters are displayed
```
const HidKeyboardKey hid_keyboard_keyset[ROW_COUNT][COLUMN_COUNT] = {
    {
        {.width = 1, .icon = &I_ButtonF1_5x8, .value = HID_KEYBOARD_F1},
        {.width = 1, .icon = &I_ButtonF2_5x8, .value = HID_KEYBOARD_F2},
        {.width = 1, .icon = &I_ButtonF3_5x8, .value = HID_KEYBOARD_F3},
        {.width = 1, .icon = &I_ButtonF4_5x8, .value = HID_KEYBOARD_F4},
        {.width = 1, .icon = &I_ButtonF5_5x8, .value = HID_KEYBOARD_F5},
        {.width = 1, .icon = &I_ButtonF6_5x8, .value = HID_KEYBOARD_F6},
        {.width = 1, .icon = &I_ButtonF7_5x8, .value = HID_KEYBOARD_F7},
        {.width = 1, .icon = &I_ButtonF8_5x8, .value = HID_KEYBOARD_F8},
        {.width = 1, .icon = &I_ButtonF9_5x8, .value = HID_KEYBOARD_F9},
        {.width = 1, .icon = &I_ButtonF10_5x8, .value = HID_KEYBOARD_F10},
        {.width = 1, .icon = &I_ButtonF11_5x8, .value = HID_KEYBOARD_F11},
        {.width = 1, .icon = &I_ButtonF12_5x8, .value = HID_KEYBOARD_F12},
    },
    {
        {.width = 1, .icon = NULL, .key = "&", .shift_key = "1", .value = HID_KEYBOARD_1},
	{.width = 1, .icon = NULL, .key = "\xE9", .shift_key = "2", .value = HID_KEYBOARD_2}, //"\xE9" = é
        {.width = 1, .icon = NULL, .key = "\"", .shift_key = "3", .value = HID_KEYBOARD_3},
        {.width = 1, .icon = NULL, .key = "'", .shift_key = "4", .value = HID_KEYBOARD_4},
        {.width = 1, .icon = NULL, .key = "(", .shift_key = "5", .value = HID_KEYBOARD_5},
        {.width = 1, .icon = NULL, .key = "-", .shift_key = "6", .value = HID_KEYBOARD_6},
	{.width = 1, .icon = NULL, .key = "\xE8", .shift_key = "7", .value = HID_KEYBOARD_7}, //"\xE8" = è
        {.width = 1, .icon = NULL, .key = "_", .shift_key = "8", .value = HID_KEYBOARD_8},	
	{.width = 1, .icon = NULL, .key = "\xE7", .shift_key = "9", .value = HID_KEYBOARD_9}, //"\xE7" = ç
        {.width = 1, .icon = NULL, .key = "\xE0", .shift_key = "0", .value = HID_KEYBOARD_0}, //"\xE0" = à		
        {.width = 2, .icon = &I_Pin_arrow_left_9x7, .value = HID_KEYBOARD_DELETE},
        {.width = 0, .value = HID_KEYBOARD_DELETE}, 
    },
    {
	{.width = 1, .icon = NULL, .key = "a", .shift_key = "A", .value = HID_KEYBOARD_Q},
        {.width = 1, .icon = NULL, .key = "z", .shift_key = "Z", .value = HID_KEYBOARD_W},
        {.width = 1, .icon = NULL, .key = "e", .shift_key = "E", .value = HID_KEYBOARD_E},
        {.width = 1, .icon = NULL, .key = "r", .shift_key = "R", .value = HID_KEYBOARD_R},
        {.width = 1, .icon = NULL, .key = "t", .shift_key = "T", .value = HID_KEYBOARD_T},
        {.width = 1, .icon = NULL, .key = "y", .shift_key = "Y", .value = HID_KEYBOARD_Y},	
        {.width = 1, .icon = NULL, .key = "u", .shift_key = "U", .value = HID_KEYBOARD_U},
        {.width = 1, .icon = NULL, .key = "i", .shift_key = "I", .value = HID_KEYBOARD_I},
        {.width = 1, .icon = NULL, .key = "o", .shift_key = "O", .value = HID_KEYBOARD_O},
        {.width = 1, .icon = NULL, .key = "p", .shift_key = "P", .value = HID_KEYBOARD_P},	
	{.width = 1, .icon = NULL, .key = "^", .shift_key = "\xA8", .value = HID_KEYBOARD_OPEN_BRACKET}, //"\xA8" = ¨
        {.width = 1, .icon = NULL, .key = "$", .shift_key = "\xA3", .value = HID_KEYBOARD_CLOSE_BRACKET}, //"\xA3" = £
		
    },	
    {
        {.width = 1, .icon = NULL, .key = "q", .shift_key = "Q", .value = HID_KEYBOARD_A},
        {.width = 1, .icon = NULL, .key = "s", .shift_key = "S", .value = HID_KEYBOARD_S},
        {.width = 1, .icon = NULL, .key = "d", .shift_key = "D", .value = HID_KEYBOARD_D},
        {.width = 1, .icon = NULL, .key = "f", .shift_key = "F", .value = HID_KEYBOARD_F},
        {.width = 1, .icon = NULL, .key = "g", .shift_key = "G", .value = HID_KEYBOARD_G},
        {.width = 1, .icon = NULL, .key = "h", .shift_key = "H", .value = HID_KEYBOARD_H},
        {.width = 1, .icon = NULL, .key = "j", .shift_key = "J", .value = HID_KEYBOARD_J},
        {.width = 1, .icon = NULL, .key = "k", .shift_key = "K", .value = HID_KEYBOARD_K},
        {.width = 1, .icon = NULL, .key = "l", .shift_key = "L", .value = HID_KEYBOARD_L},
        {.width = 1, .icon = NULL, .key = "m", .shift_key = "M", .value = HID_KEYBOARD_SEMICOLON}, 
	{.width = 2, .icon = &I_Pin_arrow_right_9x7, .value = HID_KEYBOARD_RETURN}, 
        {.width = 0, .value = HID_KEYBOARD_RETURN}, 
    },
    {
	{.width = 1, .icon = &I_Pin_arrow_up_7x9, .value = HID_KEYBOARD_L_SHIFT},
        {.width = 1, .icon = NULL, .key = "w", .shift_key = "W", .value = HID_KEYBOARD_Z},
        {.width = 1, .icon = NULL, .key = "x", .shift_key = "X", .value = HID_KEYBOARD_X},
        {.width = 1, .icon = NULL, .key = "c", .shift_key = "C", .value = HID_KEYBOARD_C},
[... 13 lines cut ...]
        {.width = 0, .value = HID_KEYBOARD_SPACEBAR},
        {.width = 0, .value = HID_KEYBOARD_SPACEBAR},
        {.width = 0, .value = HID_KEYBOARD_SPACEBAR},
	{.width = 1, .icon = NULL, .key = ")", .shift_key = "\xB0", .value = HID_KEYBOARD_MINUS}, //"\xB0" = °
	{.width = 1, .icon = NULL, .key = "=", .shift_key = "+", .value = HID_KEYBOARD_EQUAL_SIGN},	
        {.width = 1, .icon = &I_ButtonLeft_4x7, .value = HID_KEYBOARD_LEFT_ARROW},
        {.width = 1, .icon = &I_ButtonDown_7x4, .value = HID_KEYBOARD_DOWN_ARROW},
        {.width = 1, .icon = &I_ButtonRight_4x7, .value = HID_KEYBOARD_RIGHT_ARROW}, 
    },
[... cut ...]
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_L_GUI},
        {.width = 2, .icon = &I_KB_key_Tab_17x10, .value = HID_KEYBOARD_TAB},
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_TAB},
        {.width = 2, .icon = &I_KB_key_Esc_17x10, .value = HID_KEYBOARD_ESCAPE},
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_ESCAPE},
        {.width = 2, .icon = &I_KB_key_Del_17x10, .value = HID_KEYBOARD_DELETE_FORWARD},
        {.width = 0, .icon = NULL, .value = HID_KEYBOARD_DELETE_FORWARD},
    },
};
```

Willy-JL, 2024-01-28:
You discovered utf8 fonts for u8g2 exist, congrats. Doesn't change the fact that the rest of the firmware does not support utf8, especially not filenames or any other logic, and also you don't consider the fact that these fonts are bigger and consume more space on flash and possibly ram, which is very limited and already causing issues. Cool if you use it locally, not gonna apply this in the firmware however.

KaliStudio, 2024-01-28:
I understand the problem, my goal was to create the keyboard in UTF-8, I'm sharing the solution in case it helps someone else. Thank you for your time