Flipper-XFW/Xtreme-Firmware #536: Sub-Ghz Remote ERR Bad modualtion on HDN_1

lewilou22, 2024-01-27:
### Describe the bug.

When trying to make a Sub-Ghz Remote for a Honda Key Fob with HDN_1 modulation, The flipper errors out with  "ERR Bad modualtion" with HDN_1 captured .sub files 

### Reproduction

Flipper Zero with Xtreme
Open Sub-Ghz Remote 
Add new file (HND_1 recorded .sub)
ERROR BAD MODULATION 

### Logs

```Text
ERROR BAD MODULATION
```

Willy-JL, 2024-02-13:
can you share some example files perhaps?

lewilou22, 2024-02-13:
sure ! 

Here is an example of a captured HND .sub that I use to lock my car. 

The goal is to have the all the captured .sub HND files to make a remote on the flipper. 

`Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 313850000
Preset: FuriHalSubGhzPresetCustom
Custom_preset_module: CC1101
Custom_preset_data: 02 0D 0B 06 08 32 07 04 14 00 13 02 12 04 11 36 10 69 15 32 18 18 19 16 1D 91 1C 00 1B 07 20 FB 22 10 21 56 00 00 C0 00 00 00 00 00 00 00
Protocol: RAW
RAW_Data: 59 -2918 53 -1267 113 -572 69 -508 59 -794 65 -1885 79 -544 69 -754 75 -649 67 -1699 51 -128 71 -1525 106 -2906 63 -457 51 -1091 59 -380 59 -342 57 -345 53 -1477 51 -398 59 -2705 61 -437 61 -918 53 -660 66 -3018 51 -536 55 -913 67 -444 67 -3105 59 -1078 57 -633 53 -2101 87 -1798 71 -311 53 -616 59 -1225 73 -1015 61 -104 75 -256 55 -375 77 -661 57 -146 71 -126 63 -161 81 -400 51 -232 66 -72 101 -134 61 -426 79 -250 73 -108 85 -182 55 -540 71 -158 59 -754 67 -123 67 -605 67 -144 67 -158 59 -262 59 -112 83 -201 69 -518 53 -734 69 -458 71 -118 85 -677 59 -266 61 -1675 70 -501 61 -2333 51 -354 103 -338 71 -458 51 -1044 139 -671 68 -991 59 -609 75 -134 51 -625 53 -910 69 -1080 129 -118 65 -184 67 -764 71 -122 71 -116 59 -539 61 -297 61 -130 103 -531 65 -2593 73 -1368 51 -126 65 -279 53 -255 71 -371 75 -1569 61 -634 67 -104 61 -2688 71 -572 53 -418 63 -348 59 -90 67 -514 67 -1389 61 -227 73 -1046 61 -189 51 -791 89 -173 78 -144 53 -693 59 -605 61 -231 138 -286 77 -303 51 -214 66 -86 81 -634 51 -92 59 -218 61 -172 81 -138 89 -546 143 -408 77 -62 53 -82 71 -209 75 -1487 63 -561 61 -1669 69 -1167 51 -3678 75 -1058 75 -218 73 -509 51 -409 53 -337 67 -755 89 -138 71 -334 55 -212 51 -182 79 -126 61 -704 61 -533 53 -5900 53 -1124 53 -439 53 -122 69 -861 61 -375 65 -150 59 -106 51 -153 69 -72 57 -128 71 -644 63 -314 75 -102 89 -463 61 -64 61 -243 63 -108 53 -133 71 -249 73 -304 63 -128 93 -78 67 -237 71 -358 71 -616 63 -490 53 -1503 67 -561 61 -752 59 -6017 77 -86 61 -64 76 -1626 65 -209 53 -104 59 -985 53 -489 67 -494 79 -250 80 -331 69 -419 61 -358 61 -1609 67 -152 75 -168 59 -222 71 -308 63 -265 55 -313 59 -345 53 -273 75 -505 57 -232 63 -102 79 -260 67 -171 69 -174 53 -96 63 -1039 51 -2671 69 -1326 81 -232 124 -54 98 -1024 82 -106 51 -250 51 -315 79 -198 72 -152 61 -986 59 -231 59 -188 87 -310 53 -116 71 -532 93 -359 65 -148 61 -193 71 -257 51 -90 77 -268 53 -320 57 -815 77 -250 73 -494 67 -280 51 -578 65 -808 87 -4374 67 -283 67 -1370 67 -138 53 -102 75 -1353 51 -1759 107 -126 81 -1228 73 -558 51 -192 53 -1688 63 -1510 423 -228 251 -258 245 -250 251 -248 259 -246 255 -256
RAW_Data: 247 -250 251 -248 259 -244 253 -256 255 -250 247 -250 251 -260 253 -250 253 -248 243 -256 253 -246 255 -258 249 -254 239 -260 247 -256 255 -254 247 -250 251 -248 257 -246 255 -256 247 -258 243 -254 257 -252 245 -256 255 -248 241 -256 255 -252 255 -248 253 -256 247 -252 253 -248 259 -244 255 -248 249 -254 247 -260 245 -256 249 -256 249 -248 255 -252 257 -248 257 -248 241 -256 249 -252 255 -254 255 -248 249 -254 249 -258 245 -256 253 -248 251 -246 255 -252 257 -246 255 -254 247 -250 253 -254 249 -252 255 -250 249 -254 251 -248 259 -252 245 -256 249 -250 253 -246 261 -246 255 -248 247 -252 259 -254 247 -258 245 -256 247 -248 251 -256 251 -258 249 -248 247 -250 259 -246 257 -252 249 -258 247 -252 253 -248 259 -244 257 -248 249 -254 251 -252 253 -250 251 -256 247 -252 251 -256 251 -248 255 -246 249 -252 255 -258 249 -248 255 -248 247 -254 259 -246 257 -252 249 -254 245 -254 251 -254 251 -254 249 -250 247 -256 249 -254 253 -258 251 -248 247 -250 251 -254 253 -258 249 -256 237 -256 257 -250 253 -248 259 -244 249 -250 251 -256 257 -250 247 -256 249 -250 253 -248 257 -250 255 -248 245 -254 249 -258 251 -254 249 -256 247 -256 245 -256 255 -248 257 -242 247 -260 245 -256 255 -250 255 -252 245 -254 247 -252 249 -256 255 -248 241 -256 257 -244 253 -256 255 -246 247 -254 253 -250 249 -256 247 -250 251 -248 257 -248 255 -248 253 -250 251 -248 257 -250 253 -256 245 -250 251 -248 251 -252 257 -248 257 -252 245 -250 259 -248 245 -256 255 -252 245 -250 249 -256 253 -248 253 -250 251 -246 253 -252 257 -252 245 -256 247 -250 253 -246 259 -246 255 -256 247 -250 255 -248 249 -250 257 -248 249 -254 251 -246 259 -244 255 -256 249 -252 245 -258 249 -256 247 -248 249 -254 247 -258 243 -256 255 -256 239 -252 255 -250 251 -256 253 -246 245 -254 253 -250 251 -256 247 -258 241 -254 257 -244 253 -256 247 -250 251 -246 257 -250 257 -248 257 -252 245 -250 249 -256 249 -258 247 -252 245 -250 259 -248 247 -258 249 -252 247 -252 251 -256 247 -258 251 -246 247 -250 255 -248 257 -248 253 -252 247 -252 251 -256 249 -256 253 -242 247 -258 247 -256 255 -246 255 -250 251 -246 257 -252 245 -256 253 -246 247 -252 253 -258 249 -256 245 -256 245 -254 249 -254 247 -260 245 -254 247 -250 251 -252 255 -250 251 -256 247 -252 253 -248 251 -250 255 -246 247 -252 255 -250 251 -256 247 -258 241 -254 253 -250 257 -248
RAW_Data: 253 -248 245 -252 257 -252 249 -252 251 -252 247 -252 251 -258 247 -256 253 -252 245 -250 247 -256 249 -256 253 -250 251 -246 251 -250 259 -246 255 -250 249 -254 247 -252 255 -248 255 -248 245 -262 249 -252 245 -256 251 -248 247 -250 261 -246 257 -250 247 -256 249 -250 255 -248 251 -250 253 -248 247 -252 259 -246 251 -258 249 -256 239 -256 253 -246 261 -248 251 -250 251 -246 251 -256 251 -256 247 -256 245 -256 249 -246 255 -254 247 -256 239 -260 253 -246 255 -252 247 -256 247 -250 251 -256 249 -250 255 -248 253 -252 251 -254 247 -258 243 -256 499 -498 503 -498 495 -502 501 -494 249 -246  [... cut ...]  -3003 71 -186 51 -1202 53 -150 53 -253 59 -543 71 -298 61 -152 85 -466 85 -435 55 -64 59 -153 53 -180 73 -1104 75 -999 61 -162 63 -262 77 -626 86 -216 73 -395 57 -594 69 -514 79 -1761 61 -1815 69 -374 67 -196 63 -264 55 -328 67 -212 53 -282 69 -1151 65 -604 69 -555 67 -122 69 -307 69 -462 51 -365 67 -108 65 -192 61 -256 67 -1429 63 -534 71 -160 53 -1239 61 -60 69 -131 59 -676 67 -1648 79 -914 63 -2303 79 -1903 71 -302 63 -238 51 -1432 67 -814 53 -356 73 -588 65 -98 67 -1569 67 -451 51 -387 53
`

Willy-JL, 2024-02-14:
should manage to send it now, however i cant test it properly as that file you sent got mangled, you should have just uploaded the file instead of pasting the contents.

lewilou22, 2024-02-14:
[https://flpr.app/sf/#path=subghz/Honda_crv_lock_multi.sub&key=HxeN-ykK1hLFQBsIBfBUvA&id=BsjWCe](url)

Sorry about that, was on my phone at the time it was not letting me upload the .sub  I have uploaded it here. Let me know if you need me to re upload.

Thank you .