Flipper-XFW/Xtreme-Firmware #548: New update causes crash

zcbml88, 2024-02-09:
### Describe the bug.

When I open subghz or frequency analyzer, flipper crashes and says "flipper crashed but has been rebooted applications/services/expansion/expansion.c" and reboots then crashes again, then sometimes reboots with all glithed out screen saying click back button for factory reset. 

### Reproduction

Open subghz folder, then open frequency analyzer, scroll up,up,down,down,down,down.

Willy-JL, 2024-02-09:
known issue, being worked on in flipperdevices/flipperzero-firmware#3429. for now, set `Settings > Expansion Modules > Listen UART = None` and `Settings > System > Log Level = None` to fix this.

for context, the issue is copious amounts of serial feedback from a connected UART based board confusing the expansion module listener. disconnecting your wifi module (or whatever else it is) would equally fix the issue.

Willy-JL, 2024-02-09:
as for
> then sometimes reboots with all glithed out screen saying click back button for factory reset.

that is something weird youre doing, pressing random buttons.

Willy-JL, 2024-02-09:
fully fixed as of merging the above mentioned PR via  fe22543f19b13bf3fbd1b46b823bfa2687392f27 from what i can tell, i cannot get it to happen again in any way no matter how hard i try. reopening as this is still an issue in release, will close when the fix is in a public release. for now, try the dev builds (i will issue one soon, have to change a few more things first)