Flipper-XFW/Xtreme-Firmware #549: App: Geiger Counter in XFW-0053_02022024

braw, 2024-02-10:
### Describe the bug.

Geiger Counter app crashes Flipper. Using [FlippenHeimer](https://www.tindie.com/products/eried/flippenheimer-geiger-counter-for-the-flipper-zero/) board. Worked in XFW-0052_09122023, but appears broken in XFW-0053_02022024. Rolled back to XFW-0052_09122023 and have no issues.

Screen shows this after reboot:

> Flipper crashed but has been rebooted
> targets/f7/furi_hal/furi_hal_gpio.c

### Reproduction

1. Plug in FlippenHeimer board.
2. Launch Geiger Counter app.
3. System crashes and reboots.

### Target

targets/f7/furi_hal/furi_hal_gpio.c

Willy-JL, 2024-02-10:
Can you try on latest development build? I have a feeling that the expansion related changes fixed this, as I cannot reproduce it

braw, 2024-02-10:
I tried building Dev and installing. Both the Geiger Counter app and the Geiger Dice Roll still crash and reboot the Flipper. No issues when rolling back to XFW-0052_09122023. I tried formatting the SD and installing, rolling back to stock OS then to Dev XFW and it still crashes for me.

braw, 2024-02-10:
I also tried the build [here](https://github.com/Flipper-XFW/Xtreme-Firmware/actions/runs/7852314198), and get the same results. It crashes if I have the FlippenHeimer board plugged in or not.

Willy-JL, 2024-02-10:
managed to reproduce it, will have a fix soon. its related to the expansion stuff as i thought, but its just that its not been adapted to support them existing

Willy-JL, 2024-02-10:
for now can just set Settings > Expansion Modules > Listen UART = None

Willy-JL, 2024-02-10:
this is quite confusing... it crashes at
```c
furi_hal_gpio_add_int_callback(&gpio_ext_pa7, gpiocallback, event_queue);
```
which looks fine as code... the reason in crashes i see in debugger is that `&gpio_ext_pa7` already has a callback configured, but that doesnt make much sense... ill keep investigating

braw, 2024-02-10:
I appreciate you looking into it. Just wanted to report it.

Willy-JL, 2024-02-10:
this seems to be a problem with gpio pins callback from ofw. it stores interrupt callbacks based on pin index but not port. so pin 7 of right half of gpio had a callback, and setting a callback for pin 7 of left half conflicted, even though it should not have. for now, i can simply disable expansion modules in this app, but i mentioned this to ofw, hope it will get a proper fix

Willy-JL, 2024-02-11:
temporarily fixed via b9594f9b4fe92752d6bb8ed2235a395c9b6a53bf, lets hope we get a proper fix from upstream

Willy-JL, 2024-02-12:
drzlo and skotopes explained the issue, seems like it cannot be fixed properly. gpio pin interrupts are done via exti lines, which are shared between gpio ports, so pin 7 of port a will share the interrupt with pin 7 of port b and all other ports... so yeah disabling expansion module is the only option, this is a hardware limitation of the cpu of flipper.