Flipper-XFW/Xtreme-Firmware #550: Sub-ghz: not available

LinkseRat, 2024-02-11:
### Describe the bug.

I installed the new update and at first everything worked fine. But when i tried to open the sub-ghz tab again, the FZ crashed. 
I restarted the FZ and i can use everything except the sub-ghz app. After that first try it's unusable

### Reproduction

1. update the firmware
2. restart FZ
3. open sub-ghz
4. test features
5. exit sub-ghz
6. open sub-ghz
7. FZ crashes

Sil333033, 2024-02-11:
it would help if you mention what the flipper says 🤡

Sil333033, 2024-02-11:
this is already fixed in our dev branch. Update to a dev release or be patient for a new release

closed 2024-02-11 by Sil333033

Willy-JL, 2024-02-11:
Duplicate of #548