Flipper-XFW/Xtreme-Firmware #556: Crashes when quickly exiting SubGHZ Remote app.

jazkid03, 2024-02-13:
### Describe the bug.

When opening any SubGHZ remote file and rapidly pressing the back button the flipper will crash and report "Flipper crashed but has been rebooted" and "furi/core/pubsub.c".
[image]

### Reproduction

1. Power device on.
2. Navigate to and open any SubGHZ remote map file.
3. Run the file and open the SubGHZ Remote App.
4. Rapidly press the back button 2-4 times.
5. Observe crash.

### Logs

```Text
Sorry, I'm new to this and don't know where the log files are stored.
```

Willy-JL, 2024-02-13:
cant replicate. try updating to latest development build and see if it is still happening.

jazkid03, 2024-02-20:
It is, and it has been consistently happening since build release 48

jazkid03, 2024-02-20:
Also I'm currently on a work trip and only really have time to mess with this stuff on the weekends, so my apologies for the late reply.

Willy-JL, 2024-02-23:
maybe provide the files you are using and a video of how it happens. if you want we can handle this on discord if its easier