Flipper-XFW/Xtreme-Firmware #557: Error 4294967295

KaliStudio, 2024-02-13:
### Describe the bug.

I get an error when compiling :  Error 4294967295

### Reproduction

I re-downloaded the github: git clone --recursive --jobs 8 https://github.com/Flipper-XFW/Xtreme-Firmware.git
Build firmware using Flipper Build Tool: ./fbt

### Logs

```Text
./fbt COMPACT=1 DEBUG=0 updater_package

fbt: warning: Failed parsing manifest 'C:\Xtreme-Firmware\applications\external\mifare_nested\application.fam' : Invalid appid '.mifare_nested'. Must match regex 're.compile('^[a-z0-9_]+$')'
LoadAppManifest, line 132, in file "C:\Xtreme-Firmware\scripts\fbt_tools\fbt_apps.py"

fbt: warning: Failed parsing manifest 'C:\Xtreme-Firmware\applications\external\mifare_nested\application.fam' : Invalid appid '.mifare_nested'. Must match regex 're.compile('^[a-z0-9_]+$')'
LoadAppManifest, line 132, in file "C:\Xtreme-Firmware\scripts\fbt_tools\fbt_apps.py"
        PROTO   assets\protobuf\application.proto
        CC      applications\main\subghz\scenes\subghz_scene_save_name.c
        CC      applications\main\subghz\scenes\subghz_scene_save_success.c
        CC      applications\main\subghz\scenes\subghz_scene_saved.c
        CC      applications\main\subghz\scenes\subghz_scene_saved_menu.c
        CC      applications\main\subghz\scenes\subghz_scene_saved_show_gps.c
        CC      applications\main\subghz\scenes\subghz_scene_set_cnt.c
        CC      applications\main\subghz\scenes\subghz_scene_set_fix.c
        CC      applications\main\subghz\scenes\subghz_scene_set_seed.c
        CC      applications\main\subghz\scenes\subghz_scene_set_type.c
        CC      applications\main\subghz\scenes\subghz_scene_show_error.c
        CC      applications\main\subghz\scenes\subghz_scene_show_error_sub.c
        CC      applications\main\subghz\scenes\subghz_scene_show_gps.c
        CC      applications\main\subghz\scenes\subghz_scene_start.c
        CC      applications\main\subghz\scenes\subghz_scene_transmitter.c
scons: *** [build\f7-firmware-C\assets\compiled\application.pb.c] Error 4294967295
        CC      applications\main\subghz\views\receiver.c

********** FBT ERRORS **********
build\f7-firmware-C\assets\compiled\application.pb.c: Error 4294967295
PS C:\Xtreme-Firmware>
```

Willy-JL, 2024-02-13:
are you sure you cloned again? that looks like an error from un-clean build environment, after a change from upstream toolchain

KaliStudio, 2024-02-13:
I just re-cloned it now:
```

PS C:\> git clone --recursive --jobs 8 https://github.com/Flipper-XFW/Xtreme-Firmware.git
Cloning into 'Xtreme-Firmware'...
remote: Enumerating objects: 144296, done.
remote: Counting objects: 100% (4161/4161), done.
remote: Compressing objects: 100% (1345/1345), done.
remote: Total 144296 (delta 2882), reused 4021 (delta 2754), pack-reused 140135
Receiving objects: 100% (144296/144296), 596.69 MiB | 36.88 MiB/s, done.
Resolving deltas: 100% (97466/97466), done.
Updating files: 100% (4182/4182), done.
Submodule 'applications/external' (https://github.com/Flipper-XFW/Xtreme-Apps.git) registered for path 'applications/external'
Submodule 'assets/protobuf' (https://github.com/flipperdevices/flipperzero-protobuf.git) registered for path 'assets/protobuf'
Submodule 'lib/FreeRTOS-Kernel' (https://github.com/FreeRTOS/FreeRTOS-Kernel.git) registered for path 'lib/FreeRTOS-Kernel'
Submodule 'lib/cxxheaderparser' (https://github.com/robotpy/cxxheaderparser.git) registered for path 'lib/cxxheaderparser'
Submodule 'lib/heatshrink' (https://github.com/flipperdevices/heatshrink.git) registered for path 'lib/heatshrink'
Submodule 'lib/libusb_stm32' (https://github.com/flipperdevices/libusb_stm32.git) registered for path 'lib/libusb_stm32'
Submodule 'lib/littlefs' (https://github.com/littlefs-project/littlefs.git) registered for path 'lib/littlefs'
Submodule 'lib/mbedtls' (https://github.com/Mbed-TLS/mbedtls.git) registered for path 'lib/mbedtls'
Submodule 'lib/microtar' (https://github.com/amachronic/microtar.git) registered for path 'lib/microtar'
Submodule 'lib/mlib' (https://github.com/P-p-H-d/mlib.git) registered for path 'lib/mlib'
Submodule 'lib/nanopb' (https://github.com/nanopb/nanopb.git) registered for path 'lib/nanopb'
Submodule 'lib/st_cmsis_device_wb' (https://github.com/STMicroelectronics/cmsis_device_wb) registered for path 'lib/stm32wb_cmsis'
Submodule 'lib/stm32wb_copro' (https://github.com/flipperdevices/stm32wb_copro.git) registered for path 'lib/stm32wb_copro'
Submodule 'lib/stm32wbxx_hal_driver' (https://github.com/STMicroelectronics/stm32wbxx_hal_driver) registered for path 'lib/stm32wb_hal'
Cloning into 'C:/Xtreme-Firmware/applications/external'...
remote: Enumerating objects: 88662, done.
remote: Counting objects: 100% (22941/22941), done.
remote: Compressing objects: 100% (9746/9746), done.
remote: Total 88662 (delta 13024), reused 22665 (delta 12784), pack-reused 65721
Receiving objects: 100% (88662/88662), 922.79 MiB | 25.43 MiB/s, done.
Resolving deltas: 100% (56094/56094), done.
Cloning into 'C:/Xtreme-Firmware/assets/protobuf'...
remote: Enumerating objects: 290, done.
remote: Counting objects: 100% (174/174), done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 290 (delta 124), reused 127 (delta 89), pack-reused 116
Receiving objects: 100% (290/290), 59.53 KiB | 2.83 MiB/s, done.
Resolving deltas: 100% (165/165), done.
Cloning into 'C:/Xtreme-Firmware/lib/heatshrink'...
remote: Enumerating objects: 393, done.
remote: Counting objects: 100% (85/85), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 393 (delta 74), reused 71 (delta 71), pack-reused 308
Receiving objects: 100% (393/393), 146.08 KiB | 3.84 MiB/s, done.
Resolving deltas: 100% (244/244), done.
Cloning into 'C:/Xtreme-Firmware/lib/cxxheaderparser'...
remote: Enumerating objects: 1066, done.
remote: Counting objects: 100% (474/474), done.
remote: Compressing objects: 100% (122/122), done.
remote: Total 1066 (delta 396), reused 393 (delta 342), pack-reused 592
Receiving objects: 100% (1066/1066), 310.70 KiB | 5.36 MiB/s, done.
Resolving deltas: 100% (704/704), done.
Cloning into 'C:/Xtreme-Firmware/lib/libusb_stm32'...
remote: Enumerating objects: 1741, done.
remote: Counting objects: 100% (331/331), done.
remote: Compressing objects: 100% (98/98), done.
remote: Total 1741 (delta 263), reused 263 (delta 232), pack-reused 1410
Receiving objects: 100% (1741/1741), 1013.96 KiB | 11.27 MiB/s, done.
Resolving deltas: 100% (1098/1098), done.
Cloning into 'C:/Xtreme-Firmware/lib/microtar'...
remote: Enumerating objects: 252, done.
remote: Counting objects: 100% (28/28), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 252 (delta 25), reused 25 (delta 25), pack-reused 224
Receiving objects: 100% (252/252), 51.85 KiB | 2.47 MiB/s, done.
Resolving deltas: 100% (143/143), done.
Cloning into 'C:/Xtreme-Firmware/lib/littlefs'...
remote: Enumerating objects: 8488, done.
[... 132 lines cut ...]
        CC      applications\main\archive\scenes\archive_scene_search.c
        CC      applications\main\archive\archive.c
2024-02-13 07:45:01,392 [INFO] Complete
2024-02-13 07:45:01,392 [INFO] Complete
        CC      applications\main\ibutton\ibutton_cli.c
        CC      applications\main\infrared\infrared_brute_force.c
        CC      applications\main\infrared\infrared_cli.c
        CC      applications\main\infrared\infrared_signal.c
Traceback (most recent call last):
  File "C:\Xtreme-Firmware\lib\nanopb\generator\nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Xtreme-Firmware\lib\nanopb\generator\nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
ModuleNotFoundError: No module named 'proto.nanopb_pb2'
        CC      applications\main\archive\views\archive_browser_view.c
scons: *** [build\f7-firmware-C\assets\compiled\application.pb.c] Error 1

********** FBT ERRORS **********
build\f7-firmware-C\assets\compiled\application.pb.c: Error 1
```

vollukas, 2024-02-13:
I have same issue in Windows 10, but slightly different errors 
[image]
[image]
[image]
[image]

Willy-JL, 2024-02-13:
@vollukas issue has been acknowledged by hedger from ofw, will be fixed soon by them.

as for @KaliStudio's, im not entirely sure. i see you are the same person that reported the toolchain not supported issue to ofw, but that is a different error than what you reported to us here. i hope that the protobuf issue you have here is related to the toolchain not supported one that is being worked on.

KaliStudio, 2024-02-14:
Bonjour, oui, j'ai eu différents problèmes et j'ai posté aux endroits respectifs d'où venait le problème.

> @vollukasLe problème a été reconnu par hedger d'Oww et sera bientôt résolu par eux.
> 
> pour ce qui est de@KaliStudio's, je ne suis pas entièrement sûr. Je vois que vous êtes la même personne qui a signalé le problème de la chaîne d'outils non prise en charge, mais il s'agit d'une erreur différente de celle que vous nous avez signalée ici. J'espère que le problème de protobuf que vous avez ici est lié à la chaîne d'outils non prise en charge sur laquelle vous travaillez.

closed 2024-02-16 by KaliStudio

Willy-JL, 2024-02-23:
its fixed in dev branch

eried, 2024-02-23:
oh, strange the bug is back on latest dev on flipper-devices unmodded firmware (for video game module)

Willy-JL, 2024-02-23:
our dev branch includes the work in progress fix for this issue flipperdevices/flipperzero-firmware#3451, which is not yet merged into ofw