Flipper-XFW/Xtreme-Firmware #558: Video Game Module App Freeze

AbeNaws, 2024-02-14:
### Describe the bug.

When trying to run the new Flipper Zero Video Game Module app, the flipper freezes on the loading screen and becomes unresponsive. 
Disclaimer:  I do not have the module itself yet and that might be part of the problem.

### Reproduction

1. Install the Video Game Module Tool app from the app hub
2. Try to run the app on Flipper Zero (under Apps -> Tools -> Video Game Module Tool)
3. Flipper freezes and becomes unresponsive

Willy-JL, 2024-02-14:
can you try on latest dev build? man yexpansio nrelated issues in v53 that are fixed in dev

AbeNaws, 2024-02-14:
I am running XFW-0053_02022024 currently, I will try with the latest dev firmware though.

AbeNaws, 2024-02-14:
Running the dev firmware the app works fine, thanks!

Willy-JL, 2024-02-14:
thanks for checking. ill leave this open as release pending then, so others can see it until a release is out