Flipper-XFW/Xtreme-Firmware #563: Crashes When Accessing Sub-GHz After Using Marauder

AbdessalamAithaqi, 2024-02-18:
### Describe the bug.

II am encountering a recurring issue with my Flipper device where it crashes and reboots when I attempt to access the Sub-GHz application after having previously used the Marauder application. The crash displays the error message: `` flipper crashed but has been rebooted applications/services/expansion/expansion.c `` and then it reboots. The issue persists, causing the device to crash again if I return to the Sub-GHz application.

This issue is similar to the previously reported (and resolved) Issue #548 , but it seems the problem has reemerged with similar symptoms. I have the latest xtreme firmware: ``XFW-0053 02-02-2024``

To temporarily circumvent this issue, I have made the following adjustments:
    `Settings > Expansion Modules > Listen UART = None`

### Reproduction

1. Switch the flipper device on
2. Navigate to Wifi Marauder (apps>wifi>[es32]Wifi Marauder
3. Launch a scan
4. Exit the scan
5. Navigate to Subghz
6. Watch it crash and reboot in a loop

Willy-JL, 2024-02-18:
duplicate of #548

Willy-JL, 2024-02-18:
just saw this

> This issue is similar to the previously reported (and resolved) Issue https://github.com/Flipper-XFW/Xtreme-Firmware/issues/548 , but it seems the problem has reemerged with similar symptoms. I have the latest xtreme firmware: XFW-0053 02-02-2024

no. it is the same issue. the fix is not in v53. the fix is in dev branch, not in release yet. hence the issue being open and marked as "release-pending"

closed 2024-02-18 by Willy-JL

AbdessalamAithaqi, 2024-02-22:
> just saw this
>
>
> no. it is the same issue. the fix is not in v53. the fix is in dev branch, not in release yet. hence the issue being open and marked as "release-pending"

I see. Thank you for your time and your hard work. Looking forward to the next update!

cheers!