Flipper-XFW/Xtreme-Firmware #565: Battery runs out with 50%

ZhengLinLei, 2024-02-21:
### Describe the bug.

I recently noticed that my Flipper's battery runs out even though there is still battery power and sometimes it freezes before turning off. It turns on correctly and in the GUI it shows that there is a battery, but after a few minutes it turns off. I tried charging it a lot but nothing works.

### Reproduction

GUI Interface show 30% - 50% of power left but the flipper shudowns

### Target

Battery

Willy-JL, 2024-02-21:
Settings > power > auto shutdown, disable it

JoZoNoX, 2024-04-18:
I have now encountered a similar problem several times.
Unfortunately, it is also very difficult for me to recognize a pattern in the crashes/battery failures. 
However, it does not seem to be a hardware problem, as it occurs on at least two different devices (both with the latest version).

I probably won't be able to add much value, but I wanted to at least try to help in some way. What other information could I provide?
Can I somehow read out the logs afterwards and make them available?
Since the error is still occurring randomly for me, I can't narrow it down or reproduce it consistently yet. 

For me, the problem usually manifests itself as follows:
- Device is regularly switched on in the morning and switched off in the evening over several days.
- At a certain point it turns on and shows the message that the battery is completely discharged.
However, if you don't shut down but press cancel, the battery status is the same as the day before. The Flipper continues to run as usual. 
- I have also followed the advice given and deactivated the auto shutdown in the power settings.
Unfortunately, however, the battery is then suddenly so empty that the Flipper can no longer even be started, it is completely dead.

ZhengLinLei, 2024-04-18:
I fixed it by flashing the official firmware first and then the master branch of xtreme firmware. And it worked for me.

JoZoNoX, 2024-04-18:
Thanks for the reply!
I will try it