Flipper-XFW/Xtreme-Firmware #566: Flipper crashes when trying to watch videos on video player

dillkill210, 2024-02-21:
### Describe the bug.

I've recently updated my flipper via the web browser after using unleased, ive got a few video files which i was testing and now i have xtreme firmware it crashes and says:

Flipper crashed but has been rebooted
targets/f7/furi_hal/furi-_hal_interrupt.o

The firmware is: unlshd-071e 01-02-2024

I hope you can help fix this error, 
Thanks

### Reproduction

1. Open video player
2. Select a video

### Target

targets/f7/furi_hal/furi-_hal_interrupt.o

dillkill210, 2024-02-21:
I just ran unleased from the updates folder and the video player works, it's the xtreme firmware that's not letting me do it.

Willy-JL, 2024-02-21:
Wrong. The app you are using is out of date. Unleashed includes the video player and they fixed it. The app version from ofw app hub is broken. If developer doesn't fix it, they will swap the listing with one from unleashed.

closed 2024-02-21 by Willy-JL