Flipper-XFW/Xtreme-Firmware #571: Issue building/flashing Firmware with Python from source

CrashKeks, 2024-03-01:
### Describe the bug.

Since i tried after the 16.02.2024 to build and flash the Firmware from source with „fbt flash_usb_full“ on my amd64 windows 10, i get the error, that i need to install Python for flashing the Firmware to Flipper. My Problem is, i already installed Python to my System and to the System Path. The Source Code/fbt(after the 16.02.2024) don`t find Python for flashing Firmware to the Flipper after building. I already tried to reinstall and repaired Installation from Python on my  Computer. I also tried to installed the new Python version to solve this Issue. But it won`t work, if i write python —version i get normal Python installed version output. The old source code before the 16.02.2024 works. All Source Code which i downloaded after the 16.02 won`t work „Python not found…“. Please help me to solve this Problem. Is this Issue maybe a Problem with the actually source Code? I would be very happy about help and thanks in advance!

### Reproduction

Downloading the source code with git clone from master-Branch, using amd64 Windows 10 computer with Python already installed, try to build and flash firmware directly via fbt flash_usb_full. Get the Error: Python not found, please install Python via microsoft store or deactivate the shortcut in …PATH/… scons

### Target

Amd64 Windows 10/Flipper Zero/Python 

### Anything else?

Please Help

DestinyofYeet, 2024-03-01:
If you installed Python from the windows store, that won't work.

Please uninstall it and go to https://python.org and install it from there.

CrashKeks, 2024-03-01:
> If you installed Python from the windows store, that won't work.
> 
> Please uninstall it and go to https://python.org and install it from there.

I only installed python from python.org, i never installed it from microsoft store. But i deleted python and reinstalled it again from python.org, it won't work. I tried again fbt flash_usb_full and get the same error again. Please help, the version from 16.02.2024 works and all the version before the 16th, but every version after the 16.02 won't work on my windows 10 amd64 computer.
Thanks in advance!
[image]

CrashKeks, 2024-03-21:
I solved this Issue. I installed with pip heatshrink2 and pillow and its working now. Thanks

closed 2024-03-21 by CrashKeks