Flipper-XFW/Xtreme-Firmware #572: MicroPython REPL

iamrecursion, 2024-03-04:
### Description of the feature you're suggesting.

It's all well and good to have a vast array of apps—which you guys ensure we do—but sometimes I find myself wanting to prototype or hack-together functionality directly on device.

It would be _awesome_ to have a [MicroPython](https://github.com/micropython/micropython) REPL that allows you to issue commands on-device, and provides access to the GPIO pins, radios, and so on.

N-coder82, 2024-03-25:
that would be hard, not enough RAM on the device to store a REPL in memory.

iamrecursion, 2024-03-25:
You'd have to hand control directly over to micropython for the time it was running, rather than trying to run it as a flipper app.

N-coder82, 2024-03-25:
Then how would it exit the REPL

iamrecursion, 2024-03-25:
Hand it back to the OS by booting. Fundamentally it's just an alternate 'kernel' running on the microcontroller. May well be out of scope for this project but I figured I'd bring it up.

N-coder82, 2024-03-26:
Hmm, interesting, do you think you could program it yourself?

iamrecursion, 2024-03-26:
Given enough time, I think so. Unfortunately I'm very time constrained at the moment so I'm unlikely to pick it up any time soon!

ClaraCrazy, 2024-03-26:
Funnily enough, i had that same idea a while ago, unfortunately i'm far too stupid to take on such a task myself, and the very small team I do have is very busy with other features and their private lives atm

iamrecursion, 2024-03-26:
Cheers for the heads up @ClaraCrazy. I'll keep an eye on this one if I ever have time to poke at it.

N-coder82, 2024-03-26:
woo! i would love it if this made it into xtreme