Flipper-XFW/Xtreme-Firmware #573: Cross Remote fails to load due to "missing imports" on freshly updated firmware

mssevov18, 2024-03-06:
### Describe the bug.

Cross Remote (Infrared/xremote.fap) is saying "Update Application/Firmware to use this (MissingImports). I just updated to the "XFW-0053_02022024" version.

### Reproduction

1. Switch on
2. Go to apps
3. Open Cross Remote
4. Wait for it to load
5. See the error message described above
6. Only option is to go back

Tesla-7, 2024-06-01:
Having the same issue but with badbt

Frenchkyree, 2024-06-02:
Same thing for when you try to run a new command chain.

1. Open cross remote 
2. New command chain 
3. Choose + 
4. Choose infrared 
5. Flipper crashed but has been rebooted. Null pointer deference.