Flipper-XFW/Xtreme-Firmware #580: Media/Flizzer Tracker make Device reboot

TimoSEL, 2024-03-11:
### Describe the bug.

Error after reboot:
targets/f7/furi_hal/furi_hal_interruppt.c

### Reproduction

Start Flizzer Tracker

Touhottaja, 2024-03-12:
Re-producable on XFW-0053 02-02-2024.

Here's a log file of the crash (navigated from apps -> media -> flizzer tracker) if someone gets to this before me:
```
>: log trace
Current log level: trace
Use <log ?> to list available log levels
Press CTRL+C to stop...
9051 [D][BrowserWorker] Start
9064 [T][StorageApi] File/Dir 00010790 alloc
9069 [T][StorageApi] Dir 00010790 - 0001085C open (/ext/apps)
9086 [T][StorageApi] Dir 00010790 - 0001085C closed
9091 [T][StorageApi] File/Dir 00010790 free
9096 [D][BrowserWorker] Enter folder: /ext/apps items: 12 idx: -1
9103 [D][BrowserWorker] Load offset: 0 cnt: 50
9108 [T][StorageApi] File/Dir 00010790 alloc
9113 [T][StorageApi] Dir 00010790 - 0001085C open (/ext/apps)
9204 [T][StorageApi] Dir 00010790 - 0001085C closed
9210 [T][StorageApi] File/Dir 00010790 free
13516 [T][StorageApi] File/Dir 0000FDF8 alloc
13521 [T][StorageApi] Dir 0000FDF8 - 000108E4 open (/ext/apps/Media)
13533 [T][StorageApi] Dir 0000FDF8 - 000108E4 closed
13538 [T][StorageApi] File/Dir 0000FDF8 free
13543 [D][BrowserWorker] Enter folder: /ext/apps/Media items: 10 idx: -1
13550 [D][BrowserWorker] Load offset: 0 cnt: 50
13555 [T][StorageApi] File/Dir 0000FDC8 alloc
13561 [T][StorageApi] Dir 0000FDC8 - 0001087C open (/ext/apps/Media)
13569 [T][StorageApi] File/Dir 000108E0 alloc
13577 [T][StorageApi] File 000108E0 - 00010CFC open (/ext/apps/Media/bpm_tapper.fap)
13598 [T][StorageApi] File 000108E0 - 00010CFC closed
13603 [T][StorageApi] File/Dir 000108E0 free
13610 [T][StorageApi] File/Dir 00010C58 alloc
13617 [T][StorageApi] File 00010C58 - 00010D9C open (/ext/apps/Media/etch.fap)
13635 [T][StorageApi] File 00010C58 - 00010D9C closed               
13640 [T][StorageApi] File/Dir 00010C58 free                        
13647 [T][StorageApi] File/Dir 000108C8 alloc                       
13658 [T][StorageApi] File 000108C8 - 00010E64 open (/ext/apps/Media/flizzer_tracker.fap)
13688 [T][StorageApi] File 000108C8 - 00010E64 closed               
13693 [T][StorageApi] File/Dir 000108C8 free
13699 [T][StorageApi] File/Dir 00010DD8 alloc
13707 [T][StorageApi] File 00010DD8 - 00010F24 open (/ext/apps/Media/metronome.fap)
13725 [T][StorageApi] File 00010DD8 - 00010F24 closed
13730 [T][StorageApi] File/Dir 00010DD8 free
13738 [T][StorageApi] File/Dir 00010E80 alloc
13750 [T][StorageApi] File 00010E80 - 00010FCC open (/ext/apps/Media/morse_code.fap)
13780 [T][StorageApi] File 00010E80 - 00010FCC closed
13785 [T][StorageApi] File/Dir 00010E80 free
13795 [T][StorageApi] File/Dir 00010CA0 alloc
13801 [T][StorageApi] File 00010CA0 - 0001107C open (/ext/apps/Media/music_player.fap)
13820 [T][StorageApi] File 00010CA0 - 0001107C closed
13825 [T][StorageApi] File/Dir 00010CA0 free
13832 [T][StorageApi] File/Dir 00010FF0 alloc
13843 [T][StorageApi] File 00010FF0 - 00011134 open (/ext/apps/Media/paint.fap)
13859 [T][StorageApi] File 00010FF0 - 00011134 closed
13864 [T][StorageApi] File/Dir 00010FF0 free
13871 [T][StorageApi] File/Dir 00011088 alloc
13883 [T][StorageApi] File 00011088 - 000111C4 open (/ext/apps/Media/text2sam.fap)
13906 [T][StorageApi] File 00011088 - 000111C4 closed
13911 [T][StorageApi] File/Dir 00011088 free
13918 [T][StorageApi] File/Dir 00010F50 alloc
13930 [T][StorageApi] File 00010F50 - 0001127C open (/ext/apps/Media/tuning_fork.fap)
13947 [T][StorageApi] File 00010F50 - 0001127C closed
13953 [T][StorageApi] File/Dir 00010F50 free
13960 [T][StorageApi] File/Dir 000111F0 alloc
13967 [T][StorageApi] File 000111F0 - 0001132C open (/ext/apps/Media/wav_player.fap)
13983 [T][StorageApi] File 000111F0 - 0001132C closed
13988 [T][StorageApi] File/Dir 000111F0 free
14001 [T][StorageApi] Dir 0000FDC8 - 0001087C closed
14008 [T][StorageApi] File/Dir 0000FDC8 free
16813 [D][BrowserWorker] End
16829 [T][StorageApi] File/Dir 0000BE60 alloc
16833 [I][Loader] Loading /ext/apps/Media/flizzer_tracker.fap
16841 [T][StorageApi] File 0000BE60 - 0000F514 open (/ext/apps/Media/flizzer_tracker.fap)
16947 [I][Loader] Mapping
16952 [I][Elf] Total size of loaded sections: 37434
16958 [T][StorageApi] File 0000BE60 - 0000F514 closed
16963 [T][StorageApi] File/Dir 0000BE60 free
16968 [I][Loader] Loaded in 135ms
16972 [I][Loader] Starting app
16976 [I][AnimationManager] Unload animation 'L1_Waves_128x50'
``

dav3shanahan, 2024-03-13:
MAYHEM evil portal doesn’t work