Flipper-XFW/Xtreme-Firmware #585: Allow for latest build on https://flipper-xtre.me/update/

ALEEF02, 2024-03-23:
### Describe the enhancement you're suggesting.

Good afternoon,

I would appreciate it if there was a CI/CD pipeline for XFW, so that I could install the latest development build/push from the website without having to pull the repo, update submodules, fix my environment, and then finally build and upload via the fbt. This should be a relatively small implementation since there exists some CI/CD already.

N-coder82, 2024-03-23:
This actually sounds quite interesting!

N-coder82, 2024-03-23:
but @ALEEF02 it exists on github actions

ALEEF02, 2024-03-26:
Yes, I understand that I can download the builds from GitHub and then use qFlipper, however, I do believe a website integration so that it can be uploaded straight through the webserialAPI,

> but @ALEEF02 it exists on github actions https://github.com/Flipper-XFW/Xtreme-Firmware/actions/workflows/build.yml

WatskeBart, 2024-03-28:
> Yes, I understand that I can download the builds from GitHub and then use qFlipper, however, I do believe a website integration so that it can be uploaded straight through the webserialAPI,

You can use [https://lab.flipper.net/](https://lab.flipper.net/) to upload the *.tgz from the workflow build.

Of course it would be great when [https://flipper-xtre.me/update/](https://flipper-xtre.me/update/) offers a feature to upload from a file, or directly from a workflow.