Flipper-XFW/Xtreme-Firmware #586: Does this firmware work with cc1101 module?

galacticfr, 2024-04-07:
### Description of the feature you're suggesting.

I have been looking for the right information but I am still lost. I am new to this. How do I know if the cc1101 module is working after hooking it up to flipperzero with this firmware?

WatskeBart, 2024-04-07:
I have the external CC1101 booster from [euerdesign](https://shop.euerdesign.eu/products/flipper-booster-external-rf-module-beta-v2-professional-pcb), and it works with this firmware.

It detects the external module automatically.

hotdjdave, 2024-05-13:
There used to be a setting under **Sub-GHz** > **Radio Settings** > **Module** which listed the selectable options "**Internal**" and "**External**."  **Externa**l was the way to enable and external antenna and, at least to me, ensured it was using the external antenna and in the internal one.  Now, going here only shows "Internal" and no option to select "External."  Why was this taken away?  

There should be some indication on Flipper letting you know if the internal antenna is being use or an external one is used.