Flipper-XFW/Xtreme-Firmware #590: Discord captcha.bot not giving problem

super06, 2024-04-30:
I try joining the discord but the captcha.bot doesn't give me a problem

From 4 different devices it still doesn't. I click "I am a human being" then i click answer but it just says "verify yourself This for will be submitted to captcha.bot ANSWRE*

And if I click "Click to see attachment" It just highlights the original "I am a mature human being

ClaraCrazy, 2024-04-30:
jap, works for 31k people, just not for you. ofc. 

Not a bug, fix your client.

closed 2024-04-30 by ClaraCrazy

super06, 2024-04-30:
Fix my client. I used all 3 fucking clients, web , Desktop, and stupid mobile app.

super06, 2024-04-30:
> jap, works for 31k people, just not for you. ofc.
> 
> Not a bug, fix your client.

Fix my client. I used all 3 fucking clients, web , Desktop, and stupid mobile app. If your gunna gunna close the the issue might as well help me instead of being an ass