Flipper-XFW/Xtreme-Firmware #591: jap, works for 31k people, just not for you. ofc.

super06, 2024-04-30:
jap, works for 31k people, just not for you. ofc. 

Not a bug, fix your client.

_Originally posted by @ClaraCrazy in https://github.com/Flipper-XFW/Xtreme-Firmware/issues/590#issuecomment-2084604838_

Fix my client. I used all 3 clients, web , Desktop, and stupid mobile app. If your gunna gunna close the the issue might as well help me instead of being liberal

super06, 2024-04-30:
Liberal britch

super06, 2024-04-30:
Liberal britch your like my sister and mother
Both horrible people.

ClaraCrazy, 2024-04-30:
frfr ong no cap. so liberal imma let you use your own brain

closed 2024-04-30 by ClaraCrazy

super06, 2024-04-30:
Your insecure

super06, 2024-04-30:
Let me gess your next words are "ugh, you know what. I'm not gunna use any more brain power on this"  wait you where using brain power. Never would have guessed

dav3shanahan, 2024-04-30:
You’re