Flipper-XFW/Xtreme-Firmware #597: Cant connect with the flipperApp on the iPhone.

NintendoSubzero, 2024-06-04:
### Describe the bug.

Flipper is undetectable using the bluetooth. By turning it on it wont show up on my phone.
If you use one of the BLE functions it appears.

### Reproduction

1. When flipper is on, on the upper left bluetooth icon is on.
2. Go to the flipper menu
3. go to settings
4. select bluetooth
5. should be set on ON
6. go back
7. and on the iphone (17.5.1) if you go to you bluetooth settings
8. device is undetecable

### Target

IPhone 17.5.1

### Logs

```Text
Nothing is showing on the debug logs as i couldn't find any.
```

antiman9071, 2024-06-11:
Was there any progress on this as I have the same issue

TruFoox, 2024-07-01:
downgrading firmware fixed it for me