Flipper-XFW/Xtreme-Firmware #603: "Point the remote at IR port and push the button" Which Button?

tsndr, 2024-06-23:
### Describe the enhancement you're suggesting.

Maybe the wording should be changed here to tell the user which button to press:
https://github.com/Flipper-XFW/Xtreme-Firmware/blob/aaa801b9179012076458f940cba729dfe217aa58/applications/main/infrared/scenes/infrared_scene_learn.c#L16-L17

Suggestion: 
- Point the remote at IR port and push power button
- Point the remote at IR port and push any button

kreniov, 2024-07-15:
It seems clear that you need to press a button on the remote control. There's a picture there. :)

thesicillian, 2024-07-15:
Going to be honest here. I have little to no idea why you emailed me this or what you’re talking about. -Nick

tsndr, 2024-07-15:
I just wanted to fix this broken message, but I just don't care enough do argue about it.

closed 2024-07-15 by tsndr

thesicillian, 2024-07-15:
Are you meaning to email me about this Toby? I have little to almost no idea what you are reffing to? I never emailed you about this. I’m not trying to argue with you but trying to understand this email I got from you….

kreniov, 2024-07-21:
No one has written to you in the mail